ni/niveristand-fpga-addon-custom-device #96: Error 56 in VS 2019 (Linux and PharLap)

Brian-Kindinger, 2021-05-05:
I've run into an issue where I get error 56 when deploying from two different development machines to two different target (Linux and PharLap).  It transfers the files over during deployment but before it fully deploys it errors out.

bogdanp-ni, 2021-05-06:
Hi @Brian-Kindinger,

We'll look into this issue. In order to narrow it down, I would need more information:
- just to clarify - from the same System Definition, different FPGA Addon instances are deployed to both Linux and PharLap, is that your use-case?
- is the deployment successful with only one of the Linux/PharLap targets present in the System Definition?
- is deployment successful without FPGA Addon instances?
- what release version of FPGA Addon are you using?

Brian-Kindinger, 2021-05-10:
different FPGA Addon instances are deployed to both Linux and PharLap, is that your use-case?
One development machine with the FPGA Addon in VS 2019 for a PharLap target, a completely different dev machine with the FPGA Addon in VS2019 for a RT Linux target.

is the deployment successful with only one of the Linux/PharLap targets present in the System Definition?
No.  Two different sys defs, two different dev machines, two different targets

is deployment successful without FPGA Addon instances?
Yes.

what release version of FPGA Addon are you using?
https://github.com/ni/niveristand-fpga-addon-custom-device/releases/download/20.5.0/FPGA_Addon_2019.zip

DStavilaNI, 2021-06-11:
The issue was solved after recompiling the bitfile.

closed 2021-06-11 by DStavilaNI

Brian-Kindinger, 2021-06-11:
Don't you think the solution should be to catch the error and then explain that it could be the IRQ?  Error 56 is a networking issue so it seems confusing to throw that error.