ni/niveristand-fpga-addon-custom-device #125: Error 1003 when deploying for LinuxRT 64 bit (duplicated names on vi.lib)

fmussi, 2021-06-30:
**Describe the bug**
I can compile engine build distribution but I've error 1003 when deploying

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Details**
Looking to the target CD folder, there's the llb and 2 folders for the VIs with duplicated names. These are:

[image]
each of them having these 2 files:
[image]

**Please complete the following information:**
 - OS: Linux RT 64 bit
 - Release Version: latest commit (#124) 

**Additional context**
Fix is pretty easy (e.g. adding NI_Data Type lvlib to the project items and renaming conflicting items during build) but it's sort of manual and impact user productivity

DStavilaNI, 2021-07-02:
Hey Fabio,
  
              Please make sure to have the following token added to your LabVIEW.ini file: "NI_AppBuilder_SDist_ExcludeEditTime=True". 
              
              The final build should like this (without any extra files):
                            
[image]

This should eliminate the issue that you are facing.
Sincerely yours,
Denis

DStavilaNI, 2021-12-16:
Assumed fixed

closed 2021-12-16 by DStavilaNI