ni/niveristand-fpga-addon-custom-device #155: FPGA Addon Default Values Not Being Retained

aequitz, 2022-08-02:
**Describe the bug**
Default values for the FPGA Addon custom device are not retained when re-opening VeriStand System Explorer.

**To Reproduce**
Steps to reproduce the behavior:
1. Open attached system definition file in VeriStand system explorer.
2. Change default value of Controller/Custom Devices/PXIe-7821R/Scalars/Scalar Outputs/DMA.Enable to 1.
3. Open the system definition file in a text editor to confirm that DMA.Enable's default value is 1.
4. Reopen the system definition file in VeriStand system explorer.
5. Navigate to the DMA.Enable signal. The default value is now shown as 0.

**Expected behavior**
I would expect the default values to be shown the next time the System Explorer is opened.

**Screenshots**
N/A

**Please complete the following information:**
 - OS: Linux x64
 - Release Version 20.6

**Additional context**
N/A

**Attachments**
[FPGA Addon Bug Files.zip](https://github.com/ni/niveristand-fpga-addon-custom-device/files/9242702/FPGA.Addon.Bug.Files.zip)

closed 2022-08-04 by buckd

aequitz, 2022-08-05:
Is it possible to get a new release that includes this fix?

buckd, 2022-08-05:
@aequitz I'll try to get you one Monday.

buckd, 2022-08-08:
@aequitz 
New release [v21.3.0](https://github.com/ni/niveristand-fpga-addon-custom-device/releases/tag/v21.3.0)

PR #157 (2022-08-04): Reload bitfile values on sysdef load only if bitfile has changed
- [x] This contribution adheres to [CONTRIBUTING.md](https://github.com/ni/niveristand-fpga-addon-custom-device/blob/master/CONTRIBUTING.md).

### What does this Pull Request accomplish?

On system definition load, checks if the bitfile has changed since information was last parsed. If so, reload all of the settings. If the file has not changed, don't reload information from the bitfile or change anything in the system defiintion.

### Why should this Pull Request be merged?

Fixes #155 

### What testing has been done?

Manually [... cut]